YCPRadioTelescope/YCP-RT-ControlRoom #17: Priority for multiple points of control

kkennelly, 2019-09-16:
Use Case: The client wants to control the telescope from their phone, but the control room software is telling the code to do something else. 

Question we have to answer: which one has priority?

TUllery, 2020-09-20:
The Admin phone has priority, even over the control room software. When the Admin phone starts any control of the telescope, it becomes the active scheduled task and any existing task is cancelled/paused (if the Admin gives up control, the scheduled task can resume). I don't remember if the phone app has a 'release control' button or functionality, but it needs to be added.

pdnelson, 2020-09-20:
I'm thinking we can add a 'Gain Telescope Control' command to the MiddleMan or something similarly-worded that will stop the current script and allow an admin to make whatever movements necessary. The command will store its current position before the command was called, and temporarily stop data from being gathered. Then the 'Release Control' button would set the telescope back to the saved position and resume business as usual.

Does this sound like what you're looking for?

TUllery, 2020-09-20:
The logic is already built into the system. The Admin Phone App has highest priority. When the Admin app is engaged to take over the telescope,, there is a new high priority schedule made, stopping any other scheduled observations - existing functionality I believe. When the Admin app ends control of the telescope, the appointment ends. Any Primary or Secondary scheduled observations resume. There is no need to save where the telescope is pointing because it will be recalculated when the observation restarts.

pdnelson, 2020-09-20:
Confirmed with Kyle and looked into the code and I do now see this functionality, so the issue can be closed.

closed 2020-09-20 by pdnelson