YCPRadioTelescope/YCP-RT-ControlRoom #52: Sensor Checking and Overrides

AktionP, 2019-11-13:
Need to add in the checking of all of the sensors before a script is run, to prevent any runs when situations will cause damage to telescope. As well, need to add in the overrides to these checks so that if there is a serious situation in which the telescope must move, no matter what, the clients can move the telescope to a safe position

TUllery, 2020-09-20:
How is this not already implemented? If not, make it a high priority.

pdnelson, 2020-09-20:
I was thinking the same thing when I saw this in the ice box, but it's likely that overrides needed to be implemented before this. Right now it looks like the only things being checked before a script is run are temperature thresholds. I'll set this as a priority.

pdnelson, 2020-09-20:
Upon further investigation (and confirmation with Kyle), this _has_ been implemented. I must have just been looking in the wrong place for the check.

closed 2020-09-20 by pdnelson

pdnelson, 2020-09-23:
Commit: https://github.com/YCPRadioTelescope/YCP-RT-ControlRoom/commit/79b4e11a1bbb456c32dbd2413a9c78c09595a64e