YCPRadioTelescope/YCP-RT-ControlRoom #53: Grabbing PLC packet data

AktionP, 2019-11-17:
Currently the PLC packet (as far as I'm aware) is not saving or doing anything with the packets that it is receiving. The PLC sends us info about the sensors so to see the sensor data, we have to grab it from the PLC packet. Update the PLC production driver to read that packet data and set the corresponding values in the correct file. May need to also create a new class or struct to hold this data for display in the diagnostics page.

AktionP, 2019-12-07:
We are now grabbing the packet data for the limit switches and the proximity sensors

AktionP, 2020-02-12:
We are grabbing the plc packet data from the simulation and the irl telescope.
There is more testing that is being done to fix a bug in displaying the packet data (limit switches and proximity sensors) on the diagnostics page.

We also need to continue to work with the VR team to have them send back the proper values at the correct positions.

jminor717, 2020-02-13:
this fully works in production, there is still some work that needs to be done in the simulator but i'm not sure if that work falls under the this issue

AktionP, 2020-04-03:
This is as up to date as it will be for the foreseeable future

closed 2020-04-03 by AktionP