YCPRadioTelescope/YCP-RT-ControlRoom #106: Telescope Offset

AktionP, 2020-02-19:
The telescope will not have the encoders 0 not be true north 0, thus we need to have an offset for when we move the telescope, such that a point 0 degrees is correct.

jminor717, 2020-03-10:
there will likely need to be 2 offsets, one between the 0 defined by the home sensor and the 0 of the encoders, then the offset addressed in this issue. these should be kept as 2 separate values since the offset between the telescope and north will not change once the telescope is installed. the other offset between the home and encoder will be determined at run time by the home method in the PLC. coincidentally this is one of the bits of data that i wanted to store in a local database independently of the backend

AktionP, 2020-04-03:
This may have been completed already by Jake, so we will need to get in contact with him to see what he has done, if anything.

If there has been nothing done, this issue will likely be deleted as the mechanical team is currently on hold.

jminor717, 2020-04-03:
no this is not completed since the offset will need to change once the telescope is in place. but it shouldn't be deleted so that we can track the issue until that happens.

pdnelson, 2021-04-08:
The second offset Jake mentioned is now covered in #301. #301 should be completed before this one, so that we have the encoder offset working.

closed 2021-05-04 by pdnelson