YCPRadioTelescope/YCP-RT-ControlRoom #107: Expand Script List

AktionP, 2020-02-19:
Although we have a moderately sized script list, we will need to expand it to meet the demands of our clients.

Finding out what these scripts are is going to be like pulling teeth, but don't let up

AktionP, 2020-04-03:
This is currently on hold as we do not have any more scripts to add at the current momement, but we know we will need more

AktionP, 2020-04-20:
We must modify our scripts to ensure that they follow with the updated design of the Radio Telescope (e.g. Removing the checks for azimuth limit switch/hardstop checks while keeping elevation limit switch scripts)

kkennelly, 2020-04-28:
After discussion with Todd: There are no other scripts he would like implemented at the moment. He won't know until he gets to move the real telescope. Therefore, I will do some heavy documentation on how to write these scripts and where to place them in the solution.

After discussion with Jake Minor: He does not want me to change/delete any scripts that I was planning to due to the updated design. Due to COVID-19, the slip ring hasn't been ordered yet and he wishes to push this off until the slip ring is physically tested and verified.

closed 2020-04-30 by AktionP