YCPRadioTelescope/YCP-RT-ControlRoom #137: Finalize Mobile App Connection

AktionP, 2020-03-31:
We have the basics of the mobile app connection, but we need to finalize the specifics of the message parameters.

This will require working with the mobile app team to make sure that the specifics are known and understood between both teams

kkennelly, 2020-04-14:
I have done all that I can for this on the control room side. The problem seems to exist on the side of the mobile app or aws. Mobile Team will handle this issue from here on out. As for the TCP connection in the control room app, this is complete.

kkennelly, 2020-04-14:
Within this pull request, there is a line to control the telescope's movement based on the information received on the TCP Connection. We use the count - 1 to find the telescope controller in order to use the last recently created telescope. In the future, the goal is to have the mobile app decide which telescope to use and then pass that id found in the db to the control room. We can use the id to find the telescope controller. This enables control of multiple telescopes. But, in the development phase, the control room does not point to the same db as the phone app at the moment. So, this feature will have to be implemented in the future.

closed 2020-04-14 by kkennelly