YCPRadioTelescope/YCP-RT-ControlRoom #167: Encrypt MiddleMan service with AES

pdnelson, 2020-09-14:
Right now, we're sending commands to the telescope in plaintext, unencrypted. What this will do is encrypt that text so we can't be hacked.

TUllery, 2020-09-20:
Instead of everyone implementing AES in each component, how about creating one component put into a single library that everyone uses. The only field that would have to be encrypted is the one that identifies this as a valid command. The rest of the fields can be plain text. The same new component would have the code to validated the key at the destination.

pdnelson, 2020-09-20:
So, the way I was thinking of implementing this (which I admittedly did not go into much detail in the issue) is similar to what you're describing. In the code, you would still see all of the plain-text commands that would be readable. The only thing that would change is the `String data` at the top would become `Byte[] data`, then the conversion from `Byte[]` to `String` would occur on the line directly under the `processMessage` function declaration.

I was thinking the mobile team and front end could implement it similarly: Rather than encrypt every individual command, have a `String` capture the command that needs to be sent and then convert it to `Byte[]` right before it gets sent.

I may be misunderstanding what you said; if so, I'm free all day today if you would like to talk about it over Zoom!

TUllery, 2020-09-20:
I am okay with your scenario. Is there something in the encoding that is time specific so a command sent (and recorded by the evil person) today cannot be used next week?

pdnelson, 2020-09-20:
I like the time-specific idea; that will negate the need for a key that changes on every command, so it can be more secure and easier to implement. I will add a timestamp of some sort in the encoding, and then a check in the decoding that makes sure it matches.

pdnelson, 2020-10-30:
Turns out we were already receiving a `byte[]` array of data! This means instead of doing a straight conversion from `byte[]` to `String` like we were originally doing, this conversion will be replaced by the `Decrypt` function I wrote using AES.

An additional 'verifyMessage` function is being written that will verify the timestamp and then decide whether the command is able to be used or not. The way this will be implemented depends on what the mobile team and we decide, and it is currently undergoing discussion.

closed 2020-11-09 by pdnelson