YCPRadioTelescope/YCP-RT-ControlRoom #183: User validation needed for...everything

pdnelson, 2020-10-17:
Very few text fields have user validation, so they will all need gone through and verified that they are validating user input. Currently, if nothing is entered, or something invalid is entered, the program will crash.

Some that I noticed right off the bat are on `MainForm.cs`:
- MCU IP Address
- PLC port
- MCU Port
- Weather station COM port

DanPalm5, 2021-02-11:
After investigation, the best course of action was to validate the input fields in MainForm AFTER pressing the finalize settings button. Implementation focus will now shift there. More to come

DanPalm5, 2021-02-18:
Todd has stated he would like the following changes:

- Speed slider containing expected range (as opposed to slider textbox)
- Tooltips for labels containing expected values, as opposed to a message box
- Change background color to yellow or red when an invalid value is entered and disable finalize settings
- These changes should be made to both the MainForm and RTControlForm

I will begin these changes immediately!

closed 2021-02-19 by pdnelson