YCPRadioTelescope/YCP-RT-ControlRoom #272: Update Diagnostics Form - Sensor Data tab

pdnelson, 2021-03-20:
[image]
- Remove reference to second azimuth home sensor
- Remove reference to azimuth limit switches
- Add accelerometer data
  - The dotted-line square is where the FFT visualization will be
  - Directly under, there is just enough room for a color-coded key at the bottom for x, y and z
- Rename group boxes to more-accurately describe the type of sensor data they display

### ADDITIONAL POINTS DIFFERENT FROM THE PICTURE:
- Make sure both gates are being accounted for (say "gates" instead of gate)
- Change "Current Azimuth" to "Azimuth Position"
- Say Azimuth Motor Temp and Elevation Motor Temp
- accelerometer data
  - maybe average out some data because it will be changing
  - Remove "Motor" from counterbalance
  - Don't bother with making the data update or display anything; this will be covered in a future issue

davidmchugh2000, 2021-03-21:
Gate -> Gates

davidmchugh2000, 2021-03-21:
"Key:"

Makes colors same as 276

closed 2021-03-23 by pdnelson