YCPRadioTelescope/YCP-RT-ControlRoom #295: Control room exceptions out when the connection gets ended

lgartrell, 2021-03-24:
## How to Reproduce
* send more than 1 command to the control room through the middleman
* notice how it does not send a SUCCESS or FAILURE

## Work to Do
* Step through the remote listener code to figure out where it currently expects the connection to end
* Fix the listener loop and update the middleman to respect that functionality

pdnelson, 2021-05-04:
This may be a problem with the way data is being sent through the middleman. When I sent a Middleman command through using the Sensor Network demo program, it seemed to work fine. More investigation will need to go into this to figure out what's going on. 

Regardless, the Control Room should handle the error without throwing an exception.

Liam920, 2022-08-26:
This should just require a verification that CR does not in fact crash and then this issue can be closed.