YCPRadioTelescope/YCP-RT-ControlRoom #300: Fix memory leaks in RemoteListener

pdnelson, 2021-03-26:
As it stands, the `RemoteListener` uses plenty of objects that implement `IDisposable` that are not being disposed of. One of these is `NetworkStream`, though there could be more; it is up to the assignee to investigate this.

It is important to note that just creating a new instance of an object over top of the old instance does NOT dispose of the used memory, it simply gives it a new memory address. That is the main problem with the current implementation.

tfranks31, 2022-01-05:
These requirements have changed since the `RemoteListener` refactor. I looked through all of `RemoteListener` for any objects that use `IDisposable` aside from the ones mentioned above and found no other objects needed to be disposed of. There were also no cases of creating new objects over any disposable objects.

The only thing that needs to be checked is that `IDisposable` objects (`NetworkStream` and `TcpClient`) are guaranteed to be disposed of. The only situation I see where they are not disposed of is if an exception is thrown at some point in a client thread. I recommend adding `using` statements for the `IDisposable` objects to guarantee that the objects will be disposed of if an exception disrupts the thread.

jsnyder17, 2022-02-01:
Fixed with "using" statement when initializing TcpClient and NetworkStream

closed 2022-02-01 by jsnyder17