YCPRadioTelescope/YCP-RT-ControlRoom #305: Shutdown RT occasionally freezing the application

pdnelson, 2021-03-29:
This time it isn't crashing, but it is freezing.

Click the Start RT button, then quickly click Shutdown RT, and you may never close. According to Jonathan, it's stalling at `ManagementThread.Join()`.

Look into this, and see if it can be reproduced. We will want to check that this thread is being closed properly (maybe it needs a conditional statement for ending it?), and that nothing is left running.

YCPjsteck, 2021-03-29:
I just tested the potential cause that I thought of (quickly shutting down the RT after starting it) and I can't reproduce it using that method. I definitely did have the control room stall on shutdown once though.

pdnelson, 2021-04-20:
This happens primarily after executing a movement. The reason this is happening is the same reason the `TestLifeCycle` unit test endlessly runs sometimes. Fixing that unit test will also solve this problem.

Liam920, 2022-08-26:
This is still an issue, adding for MS1

robbycw, 2022-09-19:
I tried recreating this issue by both clicking Shutdown RT after executing a movement and as soon as the shutdown button was activated. It does not appear to be a current issue.