YCPRadioTelescope/YCP-RT-ControlRoom #323: Remove obsolete encoder simulator from the Appointment Control

pdnelson, 2021-04-04:
In the Appointment Control tab of the Diagnostics Form, there exists an encoder simulation, which was set in place before we finalized the design for our absolute encoders (which are now a part of the Sensor Network). As a result, this functionality is obsolete, and should be removed.

A starting point will be to remove the entirety of the lower half of the Appointment Control, as it all relates to the simulation encoders.

After this, fix any build errors, and fix any unit tests that fail as a result of this.

Update documentation screenshots, and remove any references to this encoder simulation.

pdnelson, 2021-04-08:
I'm hesitant to remove this because it has good functionality that we may be able to use. Before removing this, talk to Todd and Professor Hake and see if they have an idea about how we could repurpose it, or if we should.

The part I think could be useful is that, this allows us to set encoder values and increase/decrease them manually. I'm not sure what this could be used for, but it sounds like it may be useful and might be adaptable to the Sensor Network. Because the encoders are no longer separate entities, however, this may not be worth combining with the SN.

andersonjackc, 2021-11-17:
Met with Todd and Professor Hake and they detailed that a good use case for this would be sending test encoder data  to the HW, For example,  good encoder data, bad encoder data, and noisy encoder data.

closed 2021-11-17 by andersonjackc