YCPRadioTelescope/YCP-RT-ControlRoom #333: Create MCU Monitoring Routine

lgartrell, 2021-04-06:
## Overview
In addition to the refactor, we need to have a standard "endpoint" for MoveOrders going through the control room. This endpoint takes in the `ushort[]` register data (this data is created by MCUManager as well) and then goes into a monitoring loop **until**:
1. The MoveOrder is finished
2. We get critical sensor data which calls for **an all stop**
3. Motor or Absolute encoders are way too far off
4. Scary accelerometer data
5. A limit switch is hit 

This is a complicated problem and needs a lot of moving pieces to successfully be implemented. 

## Work To Do:
- [ ] Create final `MCUManager.cs` method `WriteToMCUAndMonitor(ushort[] registerData)`. This should return an `MCUResult` enum, and implement the logic in the diagram below.
- [ ] Create `MCUResult.cs`. This enum should encompass the different errors we could encounter and a general success result. To get a better understanding of what this should return, a full comprehension of the MCU's registers has to happen first (Ask Patrick or Lucas for the MCU manual PDF).
- [ ] Refactor the existing error enum into `MCUResult.cs`. The existing enum already has a lot of the errors we are looking for but not all of them. 
- [ ] Retry Discussion. Right now (as of writing) we do not have a solid retry plan. We need to have a design discussion with the clients to figure out which routines they want to retry, and how they should retry. Once we finish this monitoring routine, we will be ready to implement proper retry logic. 

## Diagram of Proposed Monitoring Routine
Below is an example of what the client agreed monitoring routine should do, with the while loop making the above checks at each step.

[image]

pdnelson, 2021-04-29:
608d8fe939475bfc8760ee52e691b80e56ce187f

pdnelson, 2021-04-29:
e9e16114d17c40aafa301f349f917862f6a09cc4

closed 2021-04-30 by pdnelson