YCPRadioTelescope/YCP-RT-ControlRoom #335: Add check for absolute encoders in RTController's SensorMonitoringRoutine

pdnelson, 2021-04-06:
The absolute encoders must never get too far away from the motor encoders, otherwise the motors are skipping steps and undergoing too much stress. We want to stop the telescope immediately if this happens.

For this reason, we want to perform a check in the `RadioTelescopeController`'s `SensorMonitoringRoutine` and verify that the motor encoders are not more than X degrees away from the absolute encoders. To do this, create a `CompareMotorAndAbsoluteEncoders` function that takes in the two orientations as parameters, and returns a boolean. Be sure to create unit tests for this function.

This X value is yet to be determined, so the assignee can use 10 as a "just to start" value. Make this value a constant so we know what it is.

Liam920, 2022-03-01:
Besides all of the stuff implemented, we will have to update the testing when we update the discrepancy value (which we will have to determine through testing most likely). Currently this value is set to a much too large 10.

Liam920, 2022-03-04:
Just copying a concern that Tyler brought up for this, the elevation absolute encoder doesn't have full range of motion, depending on testing we may need to account for the dead zone that the encoder cannot read (example given was potentially stops reading at 92 degrees, and should read up to 100).

closed 2022-05-08 by tfranks31