YCPRadioTelescope/YCP-RT-ControlRoom #415: Error check jog and stopping movements (says "success" when it clearly fails)

pdnelson, 2021-05-09:
It recently came to my attention that the stop movement commands were flawed.

What's happening:
1. Start up hardware and connect CR
2. Unplug hardware
3. Attempt to jog
4. Tells you "success"

Also, when it IS plugged in, it prints out "success" way too early (it should only do that when the motors are done moving).

pdnelson, 2021-05-09:
I may have fixed both of these issues in this branch: https://github.com/YCPRadioTelescope/YCP-RT-ControlRoom/tree/stop-movement-error-checking

I cannot test this on the hardware, so this should be tested before it is merged into master. It is likely that something will need to be tweaked.

DanPalm5, 2021-09-27:
We are seeing this issue slightly in reverse - we are receiving "incorrect position" script failure messages when attempting to move. However the movement DOES succeed, and it DOES move to its correct position. More investigation is required

YCPjsteck, 2021-10-19:
The bug should now be fixed on the simulation end.

closed 2022-03-28 by tfranks31