YCPRadioTelescope/YCP-RT-ControlRoom #420: Parse Acceleration Data

davidmchugh2000, 2021-05-10:
Acceleration Data is stored in the `acceration_blob `table and needs to be extracted for admin use. `acc_blob ` stores a string for 4,000 acceleration values. Each value is separated by ~ and each acceleration object is separated by -

See Todd on how he wants this information received 

Each Accelerations Stores 
- Acceleration
- X
- Y
- Z
- location_ID

andersonjackc, 2021-09-22:
I currently am working on parsing the blobs on this branch: https://github.com/YCPRadioTelescope/YCP-RT-ControlRoom/tree/acceleration-blob

I have added a parsing function, however; in creating that I encountered a few unpredicted outcomes in the actual stored data, detailed below.  I will be meeting with David to discuss the next steps for this, in order to ensure that the correct acceleration data is being stored.

As mentioned in the original issue, the acceleration data is supposed to be blobbed together as follows
acc~x~y~z~loc_ID-

When examining the database I came across the following behavior:
[image]
which makes me think that some acceleration data points are being stored without specific fields, such as x,y,z or location ID.

I also encountered:
[image]
I am not entirely sure what would cause that.

andersonjackc, 2021-09-23:
As an update for this issue, I have changed the delimiter from a - to a $ in this commit: https://github.com/YCPRadioTelescope/YCP-RT-ControlRoom/commit/2ec88c911416ad81e5cd2033babeb4f5e940dabf

I have also discovered that the huge chunk of hyphens and + symbols is SQL attempting to beautify the output of the table, as seen in this other tables output:
[image]

The blobs of acceleration data will be parsable after merging this change.

andersonjackc, 2021-12-09:
This issue has been completed, as it was opened to create a parsing function for the existing blob, which led to the entire redesign and re-implementation of the blob.  A parsing function does now exist for the blob.

A new issue has been created for graphing this data: https://github.com/YCPRadioTelescope/YCP-RT-ControlRoom/issues/467

closed 2021-12-09 by andersonjackc