YCPRadioTelescope/YCP-RT-ControlRoom #485: Fix Hardware Disconnection Exceptions

tfranks31, 2022-02-07:
While running the simulation, I decided to see what would happen if I closed it while the CR was running. To my surprise, a flood of InvalidOperationException messages filled the console. I also tried to perform some movements to see what would happen, and some different exceptions were logged in the console.

We need to figure out and fix the cause of these exceptions. Be sure to test all different ways we interact with the PLC and MCU for any more potential disconnection exceptions.

Liam920, 2022-09-21:
I tried recreating this issue and traced this issue back to the MCU manager. It appears that we do handle these issues by outputting to the log on the diagnostics form, however I believe these InvalidOperationExceptions occur once we try to access a method that calls the MCU. Specifically, Inside of `Controllers/PLCCommunication/PLCDrivers/MCUManager/MCUManager.cs`, the method `HeartbeatMonitor` attempts to call `ReadMCURegisters` or `ConnectToModbusServer` and gets a dependency error output to the console. This dependency is strict to dotnet framework 4 and would probably give a more specific error if we updated to dotnet core 6. Regardless, this issue can most likely be solved by wrapping the code in something that handles it, however this causes no issues as far as crashing the application.

Liam920, 2022-11-07:
Merged

closed 2022-11-07 by Liam920