YCPRadioTelescope/YCP-RT-ControlRoom #508: Address PLC Timeout on Disconnection

tfranks31, 2022-03-22:
Whenever the PLC/MCU loses connection to the Control Room, the elevation motor will stop resisting the weight unbalance and let the telescope fall to stow under its own weight. We believe that the PLC/MCU times out after 10 seconds. I believe that the best way to handle this would be to disable the timeout and let the motors keep holding until power is cut to them.

This will involve changing the PLC code. Find out what more needs to be done to address this issue.

Liam920, 2022-08-26:
See #416 for an issue that could be fixed in addition to this

Liam920, 2022-10-21:
The PLC code was updated to address these issues, this has been tested on the physical hardware.

closed 2022-10-21 by Liam920